TG9541/stm8ef-modbus #36: My intention now is to make the stm8s001rs485 hardware operate normally according to the modbus protocol you introduced. How do I proceed? Is there any related video operation that is helpful to me, thank you

niuyingxian, 2021-01-28:
My intention now is to make the stm8s001rs485 hardware operate normally according to the modbus protocol you introduced. How do I proceed? Is there any related video operation that is helpful to me, thank you

TG9541, 2021-01-29:
Unfortunately there are no videos.

A good way to start is connecting a serial interface to PC5 using a diode as [described here](https://github.com/TG9541/stm8ef/wiki/STM8-eForth-Programming-Tools#using-a-serial-interface-for-2-wire-communication).

You should then download or clone the repository to a Linux console (I'm using Ubuntu 20.04, but any recent distribution should work - it's also possible to use a RaspberryPi, or even a MacOS console) and follow the [description in the README.md](https://github.com/TG9541/stm8ef-modbus#installation).

Please tell me if you can do that - else I'll try to find a different way to get you started.

Here is a binary which will give you at least a Forth console on PC5 (pin7 of an STM8S001J3F3, pin4 on the header):
 
[STM8S001J3RS485.tgz.zip](https://github.com/TG9541/stm8ef-modbus/files/5891947/STM8S001J3RS485.tgz.zip)

TG9541, 2021-01-31:
I just added board support code with build automation. Tests worked fine but for pushing default node settings I'll have to add a replacement for the C0135 S2 key.

TG9541, 2021-02-01:
@niuyingxian there is a binary [STM8S001J3RS485-forth.ihx](https://github-releases.githubusercontent.com/169134522/af8fa300-63c7-11eb-87a7-7b559b6e5822?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20210201%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210201T062803Z&X-Amz-Expires=300&X-Amz-Signature=1aba50ee7d6bfa170a7aded4a5fa36f34316da2fd0ffa956235f5df58e7f2afe&X-Amz-SignedHeaders=host&actor_id=5466977&key_id=0&repo_id=169134522&response-content-disposition=attachment%3B%20filename%3Dstm8ef-bin.zip&response-content-type=application%2Foctet-stream) in the 0.20 release. If you'd find the time to test it it I'd like to learn more about your experience. Thanks in advance!

commit f0a8e95b (2021-01-31): fixes #36 add code for STM8S001J3RS485 board
closed 2022-07-09 by TG9541