TG9541/stm8ef-modbus #41: Problems with uploading modbus files on STM8S103F3 board.

szymonk159, 2021-11-08:
I am trying to do similar project like STM8S001JRS485 with modbus but using STM8S103F3 board and I can't upload modbus files on my board. I already downloaded on my board MINDEV.ihx using stm8flash but when I am uploading files in terminal some errors appear.
[image]
And this is my hardware:
[image]
Do you have any ideas how can I fix this ? 
Thank you very much in advance.

TG9541, 2021-11-08:
@szymonk159 you're trying to upload the Forth program through an RS485 dongle - unfortunately that won't work: the Forth console would have to switch your RS485 transceiver to "RX" in order to receive anything from the PC that runs e4thcom (it can be done, I guess, but depending on the type of RS485 transceiver it requires hacking the Forth kernel). 

For a console I'd like to recommend using a simple [two-wire serial interface](https://github.com/TG9541/stm8ef/wiki/STM8S-Value-Line-Gadgets#other-target-boards) with any of the following binaries:
* the STM8L001J3 binary) from the [stm8ef-modbus releases](https://github.com/TG9541/stm8ef-modbus/releases/tag/0.20) w/ console on `PC5`
* the C0135 binary from the stm8ef-modbus releases w/ console on `PD1` 
* the SWIMCOM binary from the [STM8 eForth releases](https://github.com/TG9541/stm8ef/releases/tag/2.2.28) w/ console on `PD1`

szymonk159, 2021-11-08:
Thanks for answer. So I need to buy and use 2-wire serial interface (for example PL2303)  to use terminal and upload modbus files on my board and then I can use RS485 and TTL-UART converter to communicate with computer by modbus?

TG9541, 2021-11-08:
Yes, a standard PL2303 or CH340 USB-RS232 dongles does the trick. As indicated in the sketch [here](https://github.com/TG9541/stm8ef/wiki/STM8S-Value-Line-Gadgets#other-target-boards) a diode turns the dongle's TxD into a "open collector" output (the STM8S port pin pulls the level up).   

Other possible sources for a USB serial converter are certain old interface cables for mobile phones. It's also possible to hack the USB-RS485 dongle so that it can be used as a serial interface. If you'd like to give it a try, the schematics of one of the ubiquitous black-green adapters is [here](https://hackaday.io/project/167532-modbus-things-with-stm8-eforth/log/168474-finding-the-culprit).

TG9541, 2021-11-13:
@szymonk159 I'll close this issue for now - please reopen if you have more questions.

closed 2021-11-13 by TG9541

szymonk159, 2021-11-21:
It is me again. I sucessfully downloaded STM8S001J3.ihx as you recommended on my STM8S103F3 board. Then I wanted to upload modbus protocol from STM8S001J3RS485 file using halfduplex on PD5 pin but during compilation it stops on uploading WIPE RAM at line 18 in UARTISR file and I dont know why. Firstly I uploaded BUSCTRL and console shows ok. After #include STM8S001J3RS485/board.fs I have some problems.  Do you have any advice? Thank you very much in advance.
[image]