TG9541/stm8ef-modbus #43: change node-id and baud without console

nav208, 2022-03-02:
I have C0135 4-Relay Board.
Now i flash out/C0135/C135-forth.ihx with ST-LINK V2 and linux stm8flash.
Now have Node-ID 1 and 9600 baud.

I need change baud 19200 and Node-ID 4
If i send via USB to RS485 dongle 01 06 40 00 00 04 9D C9, then board reply 01 06 40 00 00 04 9D C9, bud Node ID is 1.
I try change baud to 19200, send 01 06 20 00 00 02 03 CB and reply is 01 06 20 00 00 02 03 CB. Baud is still 9600.

I try 01 06 00 00 00 04 88 09 bud Node-ID is not change.
How i change node id and baud without forth console?

Thanks

TG9541, 2022-03-02:
@nav208 great that it works in general! 

How did you determine the "holding register" address? 

According to the docs (see the [second table here](https://github.com/TG9541/stm8ef-modbus#supported-modbus-function-codes)) EEPROM cells start at offset 60000 (which is EA60 hex).

nav208, 2022-03-03:
I try now 01 EA 60 04 09 EF 
(01 is actual nodeID,  EA60 id MB addres for change nodeID,04 is new nodeID, which i need and 09EF id parity check.)
Reply is 01 EA 01 AE A0 and nodeID is still 1.
Can you write me please full example hex for change nodeID to 04? 
Thanks for good firmware and good work :)

TG9541, 2022-03-03:
I don't have a test setup right now (it's somewhere in the basement, to be honest). 

Maybe you should review how you generate your MODBUS strings (or use a tools like [QModMaster](https://sourceforge.net/projects/qmodmaster/).

Your old test string was this:

```
01 06 40 00 00 04 9D C9
n  FC ah al dh dl ch cl 
```
The "Function Code" is 6 (hex - FC), the "Holding Register" address is 4000 (hex - ah, al), the data is 0004 (hex - dh, dl) and then the checksum. 

Your new one is:
```
01 .. EA 60 .. 04 09 FF
n  FC ah al dh dl ch cl 
```
So the FC is missing and the high byte of the data (dh) is missing.

nav208, 2022-03-03:
now is my code 01 06 EA 60 00 04 BC 0F
01 =1 (akcual node ID)
06EA = 60000
0004 = 4
BC0F = check

Reply is 01 06 EA 60 00 04 BC 0F and nodeID is still 1 :(

i use Modbus Pool for windows.

nav208, 2022-03-03:
Change baud work:
With 01 06 EA 61 00 04 ED CF is changed baud to 19200, work fine :)

Change slave ID not working:
If i change id from 1 to 4, with code 01 06 EA 60 00 04 BC 0F nodeID is changed to 4. Relay is working, but board not send modbus respons, software then show timeout error (null response). 
If nodeID is 1 after eeprom factory reset, thne response is ok. Any idea? Thanks

TG9541, 2022-03-03:
Thanks for the test. I'll look into it.

TG9541, 2022-03-04:
I reviewed the code - and I don't see why it shouldn't reply to the changed ID (4). 

Can you confirm that the your Modbus Windows software indeed sends on ID 4 when you get a time-out? Also please make sure that you reset the board before trying to send commands on the new ID.

nav208, 2022-03-04:
Yes, after change nodeID power off - on.
My software ModbusPool send every 500ms send
Tx:000006-01 03 00 00 00 0A C5 CD
and board send reply
Rx:000007-01 03 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 A3 67

After change nodeID board no send reply (null).

nav208, 2022-03-04:
Can I set with user EEPROM (60002 .. 60319) so that 
input1 = open = relay1open
input1 = close = relay1close

Thanks

TG9541, 2022-03-04:
> Tx:000006-01 03 00 00 00 0A C5 CD
> Rx:000007-01 03 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 A3 67

That looks a lot like reading 10 (0A hex) holding registers from node 1... 

> Can I set with user EEPROM (60002 .. 60319) so that
> input1 = open = relay1open
> input1 = close = relay1close

You can do that - and with relative ease - but you'll have to program some Forth. The best we to do this is with a serial terminal program - e.g. e4thcom - using a TTL-serial USB dongle connected the programming connector.

closed 2022-07-09 by TG9541