TG9541/stm8ef-modbus #44: Implement a minimal MODBUS-RTU Client

TG9541, 2022-06-27:
A review showed that implementing a very basic "MODBUS Client" based on UARTISR. The use case is application programming oriented and is thus very similar to writing application oriented MODBUS Servers (e.g., as described in the like in the `README.md` and sketched in the [test folder](https://github.com/TG9541/stm8ef-modbus/tree/master/test)).

The software architecture shall re-use the hardware layers and add just the minimum necessary for sending FC requests to a MODBUS server and receive and error-check the response. The application is responsible for implementing the required MODBUS Client FC encoders and decoders. An example application implementing reading and writing "Holding Registers" shall be provided.

The Architecture shall be as follows:

Layer|Source file|Description
-|-|-
6|`clientapp.fs` or `{BOARD}/board.fs`|configuration and application layer
5|`clientfc.fs`|**MODBUS Client FC encoders and decoders**
4|`CLPROTO`|**MODBUS Client protocol base**
3|`UARTISR`|buffered UART communication
2|`BUSCTRL`|bus access (i.e. RS485 direction control)
1|STM8 eForth|lightweight interactive multi-tasking OS

TG9541, 2022-07-03:
Update: basic MODBUS client communication works (that's how #45 was found and fixed).

niuyingxian, 2023-11-08:
Hello, I am from China, how can I use eforth for transplantation? What do I need to prepare if I start studying? Thanks

TG9541, 2023-11-11:
> Hello, I am from China, how can I use eforth for transplantation? What do I need to prepare if I start studying? Thanks

@niuyingxian I'm not sure if I understand what you mean (I guess that you use machine translation like many of my Chinese colleagues - maybe you should rephrase it?). 

You can use the code here in any way you like. You can start with downloading a binary from the [Releases](https://github.com/TG9541/stm8ef-modbus/releases). A binary for the 4-Relay-Board is in stm8ef-bin.zip (/out/C0135/C0135-forth.ihx). You need to flash it to the board with an STM8 programmer.

If you want to build you own binary you need a Linux system (e.g., a Raspberry-Pi, Windows WSL...).